AceCentre/RelayKeys #3: Mouse control

willwade, 2019-01-21:
Needs some thought this..

My suggestion:
We provide two Forms of mouse control:

1. Mouse free-flow window

The window acts like a glidepad. You move the mouse towards the top left - it moves the mouse on the second screen in the same way towards the top left of the screen.

2. Mouse control with buttons

A window is provided with buttons like so:

[image]

Clicking on these buttons (or activating the relevant keyboard shorcut for each button) - activates the mouse. Note Continuous = Continually moves until user presses an Incremental or action button. Incremental  = Moves the cursor in small amounts (5 px at a time for example).

willwade, 2019-02-13:
NB: We can now do this over the CLI & Create the GUI in our AAC App of choice. Key thing is making the AT Commands work for mouse control https://learn.adafruit.com/introducing-bluefruit-ez-key-diy-bluetooth-hid-keyboard/sending-keys-via-serial#raw-hid-mouse-reports-8-11

hosseinamin, 2019-02-13:
Here are the cli commands.

```
relaykeys-cli mousemove:5,5 # 5 right and 5 down
relaykeys-cli mousebutton:l,click # left button click
relaykeys-cli mousebutton:r,doubleclick # right button click
```

closed 2019-02-15 by willwade