AceCentre/RelayKeys #10: Switch paired device

willwade, 2019-02-15:
A way of switching between paired device 1, 2, 3, 4 etc..

willwade, 2019-02-18:
It seems like this might not be possible (from https://forums.adafruit.com/viewtopic.php?f=8&t=147745&p=729905#p729905) “A bonded central can't tell a peripheral to ignore its bond with another central that's already connected. The peripheral won't even respond to the messages.”

RoganDawes, 2019-04-11:
I think you may be confusing who would be making the decision to switch. The Bluetooth device emitting keystrokes is the peripheral. You communicate with the peripheral over USB-Serial, not via Bluetooth. As such, you can instruct the peripheral to send key events to whichever central you choose.

As seen here (https://os.mbed.com/forum/team-63-Bluetooth-Low-Energy-community/topic/31109/?page=1#comment-58678), it is possible to have multiple simultaneous connections from different central roles. So long as you keep track of those individual "channels", you can emit events to whichever you choose to.

Sticking with the VNC protocol approach, you could choose the target of the events by logging in with a different password, or implementing a custom escape message to terminate the current VNC session with the dongle and return to "AT command mode". Then, a different central could be selected using "AT+VNC=<MAC of chosen central>", or similar.

This could be implemented on the host PC by having a VNC server on multiple ports, one per remote central, and switching between them as necessary depending on which one is receiving events. Obviously, one would still need locking, and a handshake protocol between the two or more client connections as to which one has the dongle configured for which central destination, but that is all achievable.

While there may be some latency involved in sending the escape code to terminate the VNC session with a particular central, handing over control of the dongle to another VNC server instance, etc, this is not the sort of thing that happens hundreds of times per second, it is a conscious context switch at human scale to select or look at a different computer, and the latency is unlikely to be more than that already experienced by the user.

willwade, 2020-07-08:
This is now doable - see https://github.com/AceCentre/morAce/blob/a77a2fbb6be165ec55c39d6f4cb0c262470868f6/morAce/morAce.ino#L262

willwade, 2020-09-19:
We've now done this. https://github.com/AceCentre/RelayKeys/commit/69fffd89cf5ace9ee74ed6bc4fe958bff4fb3db2

closed 2020-09-19 by willwade