AceCentre/RelayKeys #19: relaykeys-qt - buttons for LGUI, etc - so you can press the button - and capture any other keys

willwade, 2019-03-06:
so because the qt version cant capture windows key - im wondering about a button somewhere"Windows key" (LGUI) that would be like a toggle button - accessible by the keyboard.

hosseinamin, 2019-03-08:
`relaykeys-qt` is capturing `LGUI` on my machine. I have to double check.

willwade, 2022-07-06:
It does capture it - its just a pain as it also triggers all actions. We need a way of the QT app having a mode to capture all input and stop the rest of the OS reacting - I think that may be hard

willwade, 2022-07-22:
I have no idea how we do this.. I think only way around it is to make it fullscreen. I'm closing for now. Its not essential to the project focusing on AT users

closed 2022-07-22 by willwade