AceCentre/RelayKeys #20: Shift 2 = @ on Mac. Shift 2  = " on Windows. What to do?

willwade, 2019-03-07:
So - if I send LSHIFT,2 I get a @ - but actually it should be " - from a windows keyboard. What should we do about that?

hosseinamin, 2019-03-08:
Can i see an image from these keyboards? Strange.

Are you saying it should be double quote?

willwade, 2019-03-13:
Mac:
[image]
[image]

Windows: 
[image]
[image]

This may be of use: https://github.com/Mottie/Keyboard/tree/master/layouts 

(they've taken it largely from http://ascii-table.com/keyboards.php)

hosseinamin, 2019-03-13:
@willwade Thanks

hosseinamin, 2019-03-13:
What will shift + quote do, on windows?

hosseinamin, 2019-03-13:
I mean `@` sign, on the windows layout version.

willwade, 2019-03-22:
So I send try and send the shift,backtick key (@) - and I get  ~ (tilde).
If I send Shift 2 - I get @ (which is the Mac layout).

hosseinamin, 2019-03-22:
> So I send try and send the shift,backtick key (@) - and I get ~ (tilde).
> If I send Shift 2 (quote) - I get @ (which is the Mac layout).

I want to know what will show up in `relaykeys-qt` when you press with or without shift these keys (`@` and `~`) on the black keyboard in that image.

hosseinamin, 2019-05-28:
@willwade I figured out what the issue is.
The keyboard you have in your windows machine has UK keyboard layout. Still in your windows machine layout has set right. But on the host machine (say iOS) US keyboard layout is selected. I think if you change the layout on iOS it will get fixed.

Also `@` sign was in `relaykeys-qt` which I did add in my earlier commits.

joedevsys, 2022-05-12:
see #94 nonchars_key_map needs to match your operating system's keyboard localisation

closed 2022-05-17 by willwade