AceCentre/RelayKeys #21: How would drag work?

willwade, 2019-03-07:
So I can see we can emulate button press etc.. but can we do something like a drag function?

hosseinamin, 2019-03-08:
I need to test this one. I will do when the ble device arrives

hosseinamin, 2019-11-07:
Here's how I did drag and drop.
```
$ relaykeys-cli --delay 150 mousebutton:l,press mousemove:50,50 mousebutton:0
```

joedevsys, 2022-02-21:
> Here's how I did drag and drop.
> 
> ```
> $ relaykeys-cli --delay 150 mousebutton:l,press mousemove:50,50 mousebutton:0
> ```

A similar approach also work when used in separate steps (in the same way some special access joysticks work with a latching drag button):
 $ relaykeys-cli  mousebutton:l    //user presses drag button
 $ relaykeys-cli  mousemove:<x>,<y>  //user  moves mouse 
 $ relaykeys-cli  mousemove:<x>,<y>  //user moves mouse some more
...
 $ relaykeys-cli  mousebutton:0   //user presses drag button again

joedevsys, 2022-04-20:
Yes this works.
$relaykeys-cli mousebutton:L,press   //Phase 1 user activates Drag Start button
...
$ relaykeys-cli mousemove:x,y   //user moves mouse
$ relaykeys-cli mousemove:x,y   //user moves mouse some more
...
$ relaykeys-cli mousebutton:0   //user  activates Drag Stop button 

Beware that mousebutton click, doubleclick or even rightclick will clear the drag operation.

I thought this would also work as swipe operation when sent all together e.g. Swipe Right:
$relaykeys-cli mousebutton:L,press 
$ relaykeys-cli mousemove:100,0
$ relaykeys-cli mousebutton:0   
But on my android phone this is interpreted as a drag-select operation. Testing using the phone's touchscreen, swipe and drag-select are very similar but to swipe you touch and move immediately, but to select text you pause slightly before moving your finger. Does this imply the commands aren't arriving quickly enough for Android to accept the swipe and deciding it's a drag?

willwade, 2022-07-06:
Thanks Joe. Im going to close this - but I get your issue. I see something similar on iOS. They are doing wierd things when a mouse is used on those OS' for touch.

closed 2022-07-06 by willwade