AceCentre/RelayKeys #22: LMETA isn't being sent..

willwade, 2019-03-07:
So on my client computer I can go to https://keycode.info - and press the apple/windows key - and I can see its detected. 

On the machine running relaykeys I do a 

``relaykeys-cli.exe keypress:LMETA``

And although I see success on the command line - I get no input on the client

hosseinamin, 2019-03-08:
`LMETA` is not windows/apple key. Use `LGUI` instead. 

You need to add `LMETA` as modifier. Try this. `relaykeys-cli.exe keypress:NONE,LMETA`

willwade, 2019-03-08:
LGUI - sends nothing - and LMETA sends nothing sadly.. Something isnt right here. Cant figure this one out.  

(What is LMETA then if LGUI is Windows key? maybe this: https://www.computerhope.com/jargon/m/meta-key.htm)

hosseinamin, 2019-03-08:
I will check What's wrong with my new OS.
I don't know What's `LMETA`. It can be option key of mac devices.

closed 2022-07-06 by willwade