AceCentre/RelayKeys #26: Support consumer media keys.

willwade, 2019-03-13:
This is needed for iOS etc control. See AT+BLEHIDCONTROLKEY at https://learn.adafruit.com/introducing-adafruit-ble-bluetooth-low-energy-friend/ble-services and consumerKey here https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/blehidadafruit

RoganDawes, 2019-04-11:
Still trying to convince you :-)

VNC would make this simple, as you just need to detect the local key event, and make sure that there is a keyboard mapping for it on the bluetooth side.

willwade, 2022-10-17:
Initially done with https://github.com/AceCentre/RelayKeys/pull/129 - VOLUP VOLDOWN or MUTE. adding a whole new AT command avoided but if this needs it we can review.

closed 2022-10-17 by willwade