AceCentre/RelayKeys #28: Problem with i12/+ - RelayKeys stops the Grid3 seeing the GEWA module

willwade, 2019-03-26:
Plug in a adafruit nrf52840 express to a i12. Go to the Grid - notice how the GEWA module can no longer be found. Take it out - it comes back... No idea.

RoganDawes, 2019-04-10:
With no experience with this "GEWA module", and assuming it is this (http://www.spectronics.com.au/product/16917), I suspect that the problem is that both devices present a USB-Serial port when plugged in, and that the GEWA driver is getting confused as to exactly which serial port it should be opening. 

It may be possible to configure this in some way, perhaps by having the RelayKeys device enumerate as a different sort of USB device to avoid confusion, or perhaps by configuring the GEWA driver to open a more specific serial port (e.g. by USB VID:PID match, or by USB serial number).

willwade, 2019-04-11:
Thanks @RoganDawes :) The GEWA module - is a IR programmable unit - a bit like this: https://www.abilia.com/en/product/progress-star-usb?product_category=24 - they build it into some devices (e.g. https://thinksmartbox.com/product/grid-3/environment-control/). And yes - you are right. Somewhere - either their firmware isn't liking another USB-Serial device in the bus. It strikes me that they are doing some lazy checking for their own device (we check the vid/pid numbers that relate to the nrf52840). Needs some more digging. Thanks for looking into it 👍

willwade, 2020-06-01:
Solved if we can do #39 (But apparently Smartbox have a solution for this - just waiting to hear back what it is)

willwade, 2022-07-06:
Our workaround right now is to use wireless mode.

closed 2022-07-06 by willwade