AceCentre/RelayKeys #33: Need to add more maths symbols

willwade, 2019-05-07:
Symbols like Square root, Infinity, Degrees all need to be a type: command.

joedevsys, 2022-02-21:
As the project is based on HID you can only send HID key codes, therefore I think the only way to send these characters is using the target device's unicode input method.  Awkwardly it is different for each operating system (and for windows varies depending on version and some registry settings) so you need to know what your target is. 

e.g. Windows 10: LALT down,KPADD,<KPhexdigit>,<KPhexdigit>,<KPhexdigit>,<KPhexdigit>,LALT up
e.g. Linux: LCTRL+LSHIFT+u,<hexdigit>,<hexdigit>,<hexdigit>,<hexdigit>,\n

These could be sent as a command sequence from the user's application without further changes to RelayKeys, or, more usefully the cli could do the translation from unicode char to command sequence. This makes sense as you already translate some chars in the paste and type commands. 

You would need some way to know what the target OS is - I'm doubt that' obtainable over HID so would wither need in the config file or settable via the cli (perhaps then stored with connected device info).

willwade, 2022-07-06:
I need to test this but I think this is fixed by using the keymap mapping in the cli application  https://docs.acecentre.org.uk/products/v/relaykeys/developers/relaykeys-cfg#introduction