AceCentre/RelayKeys #63: New install relaykeys-qt crashes "no module PyWinhook"

joedevsys, 2021-12-28:
New install of latest pre-compiled windows installer fails to run.

Machine: Clean i13 with no previous installation of ReplayKeys.  Windows 10 21H1. 

Steps taken: 
Downloaded Windows installer "RelayKeys.setup.exe". OK.
Ran installer successfully. OK.
Ran "RelayKeys" from start button, crashed immediately with the following report: 
"Failed to execute script 'relaykeys-qt' due to unhandled exception: No Module named 'PyWinhook'
Traceback: file "relaykeys-qt.py", line 15, in <module> ModuleNotFoundError.

In Services control panel the ReplayKeys service was present but not running. On clicking start it refused to start.

willwade, 2021-12-29:
Yeah - I found that just before Christmas. I think I had compiled it and when tested it never found that problem as it was picking it up locally. I'm currently trying to rebuild it with https://www.lfd.uci.edu/~gohlke/pythonlibs/#pyhook - and looking for a replacement. PyHook is horrible since it doesn't easily compile on py 3.x. 

Thanks Joe

willwade, 2022-03-09:
Note to self

Need to use something like pynput in the future

Eg

'''

from pynput import keyboard

def on_press(key):
    try:
        print('alphanumeric key {0} pressed'.format(
            key.char))
    except AttributeError:
        print('special key {0} pressed'.format(
            key))

def on_release(key):
    print('{0} released'.format(
        key))
    if key == keyboard.Key.esc:
        # Stop listener
        return False

'''

willwade, 2022-03-30:
Fixed now we've ditched pyWinHook

closed 2022-03-30 by willwade