AceCentre/RelayKeys #72: Relay keys daemon locks up if CLI send longer texts

joedevsys, 2022-02-26:
**Describe the bug**
When sending text with the paste or type commands text strings over about 20 to 40 characters can cause a failure. Sometimes it works for a few attempts but eventually locks up. If sending via  relaykeys-cli.exe you can see the command window reports a number of successful key events and then starts timing out. Once it has started timing out it never recovers until the daemon service is restarted and the remote computer connection re-established.

I tried changing baudrate (device, cfg file & port settings together) e.g. 9600 but not convinced this helps.

**workaround**
Calling the cli with a small delay using the --delay argument seems to fix it. In my case 2mS improved things slightly but still not reliable. 5mS seems more reliable.

**System Info:**
Windows 10
Bluefruit LE Friend @115200 baud

willwade, 2022-03-30:
@joedevsys not closing this till we get a chance to test this properly - but think its done. https://github.com/AceCentre/RelayKeys/releases/tag/1.7 - just note the Arduino code too: https://github.com/AceCentre/RelayKeys/blob/1.7/arduino/arduino_nRF52840/arduino_nRF52840.ino

willwade, 2022-03-30:
On looking at this the local host bug looks to have fixed this

closed 2022-03-30 by willwade