AceCentre/RelayKeys #85: Documentation suggestion - BLE Friend configuration

joedevsys, 2022-04-05:
A useful configuration command for the BLE Friend. This should be inserted before the line "Turn on HID" so the ATZ command comes afterwards:

[Optional] Name your Friend to help you identify it when you are pairing the bluetooth connection.
        AT+GAPDEVNAME=RelayKeys    (or whatever name you want)

willwade, 2022-04-05:
I feel I probably need to write a page on BLE Friend and state that its a bit wobbly in its usage..

But @joedevsys  - watch your postbox. I'm going to send you a new board. I think that might be a quicker fix to a lot of these LEFriend issues 📮 👍

joedevsys, 2022-04-06:
After a factory reset, reconfigure, and using 38400 baud it looks much more stable.

closed 2022-04-21 by willwade