AceCentre/RelayKeys #90: Define NON-US & NON_US_NUM keys

joedevsys, 2022-04-19:
These appear to be defined as CUSTOM~ and BACKSLASH but these non-standard names could cause confusion for people familiar with NON-US & NON_US_NUM, and for people doing other localisations. (Actually it's all confusing to the uninitiated anyway!) Even though I'm familiar with key names I did find BACKSLASH helpful, so could add NON-US & NON_US_NUM as aliases?

willwade, 2022-04-20:
Joe. Now I’m being stupid. What keys are these ?! 

Nb. Joe is referring to https://docs.acecentre.org.uk/products/v/relaykeys/developers/reference-2#command-keypress-key-modifier (see table with CUSTOM) in it b

joedevsys, 2022-04-20:
0x32	Key_NON_US_NUM   is used for  # and ~  in UK layouts,  relaykeys docs appears to define this as CUSTOM~
0x64	Key_NON-US             is used for \ and | in UK layouts , relaykeys docs appears to define this as BACKSLASH

These two keys can produce different characters in other language layouts, the docs define them with a UK interpretation which could confuse someone wanting to use relaykeys for a different language.

I wanted to send # but couldn't find # or HASH in the doc so therefore looked for the standard key name for this key NON_US_NUM which wasn't defined either. Eventually I worked out it has been defined as CUSTOM~  which was unexpected.

closed 2022-04-20 by willwade