AceCentre/RelayKeys #94: nonchars_key_map needs to be localisable

joedevsys, 2022-05-12:
nonchars_key_map seems to be a US mapping. It needs to be localisable for different languages. 
e.g  In UK localisations:
- Can't type or paste '£' or '\' in UK language environments.
- type:£ produces nothing. £ should be mapped to ("3", ["LSHIFT"]) but # is mapped to this (command type:# produces £ instead)
- type:# produces £. It should be mapped to NONUSHASH not ("3", ["LSHIFT"])
- type:\ produces nothing. It should be mapped to NON-US-BACKSLASH (0x64) but it's mapped to "BACKSLASH" which is the US keycode.
- NON-US-BACKSLASH (0x64) is not defined in blehid.py keymap 

Not sure of the best way to do this.  Either have multiple nonchars_key_maps and switch between them depending on a language setting, or read keymap from a file. The latter would allow translators to contribute language map files and the one to be used defined in the config file.

willwade, 2022-05-16:
This looks fixed in this PR https://github.com/AceCentre/RelayKeys/pull/95 - Thanks both :-)

willwade, 2022-05-17:
fyi - docs on this at https://docs.acecentre.org.uk/products/v/relaykeys/developers/relaykeys-cfg and https://docs.acecentre.org.uk/products/v/relaykeys/developers/reference-2#defining-a-keymap-c 

Note too latest release has all this now added