AceCentre/RelayKeys #105: Tweaks to QT

willwade, 2022-07-07:
Its a bit ugly. And lacking a few things to make it more useful:

- [ ] The large area at the bottom was originally designed to be a mouse capture area. We either try that again or drop it. 
- [ ] Deal with https://github.com/AceCentre/RelayKeys/issues/19#issuecomment-1176139162 (related to above). Maybe we need to make it a fullscreen app. Or maybe we need to try and create a way of locking focus of the app. 
- [ ] Add record macro function. On record it will record key and mouse movements as a macro file. User can then save that or just replay a macro file
- [ ] Wireless mode - we have a problem that the daemon is default running in wired mode. Can QT be a way of turning on/off the wireless mode on the daemon? 

any other ideas?

The main purpose of QT is to demonstrate functionality without digging into the CLI or Serial API - but some may find it useful as a tool to setup and configure - and use generally. It needs a bit of love in general

joedevsys, 2022-07-07:
- The layout is not great for eyegaze users.
- Current BLE device name should be a label or text box not looking like a button, preferably at the top or else at the bottom like a status bar.
- There's no display of the device list without going to delete which feels a bit dangerous.
- As some voca software can't accept text returned from a cli call they can't display the current ble device name after calling ble_cmd:switch. I have had to resort to popping up the QT app and killing it on a timer (yuk!). I'm toying with the idea of an simple alternate QT app that only displays the device name and no more as a pop-up which self-dismisses after a short time (possibly with delay time as a parameter?).

joedevsys, 2022-07-07:
The qt has keycode tables. It should use the cli_keymap

willwade, 2022-07-07:
Ok downside is a user has to turn off a default windows setting for full screen apps under focus assist:
[image]

But looks to be working fine. - but give it a go Joe
[image]

willwade, 2022-07-11:
I was looking at the bow discontinued keyshare manual and it got me thinking that maybe we need an indicator light for whether it's connected or not. In its simplest state it just shows a pairing/paired status and a connected status. 

More info at http://iogear.custhelp.com/app/answers/detail/a_id/2583

Some key pics attached [image]

[image]

[image]

willwade, 2022-07-12:
Here's my latest idea. 

- 4 big tabs down left or on top. Capture , Macro, devices ,connection 
- capture : toggle switches for capture keyboard & another for capture mouse. Indicator icon for connected to board. Indicator icon for connected to a device.  Label to show what device it's connected to. Switch device button. Last sent characters field at bottom. 
- macro tab. Our macro buttons. Record , play etc 
- devices tab. List of devices. Remove selected. Remove all. Add a device. 
- relaykeys/ connection (?). Connected indicator. Reconnect. Wireless mode on / off etc. 

Couple of shoddy pics attached [image]

[image]

joedevsys, 2022-07-14:
> I do think it could benefit from https://github.com/ms7m/notify-py for

i13 eyegaze user confirms you cannot dismiss Windows Toast popups. I haven't been able to to compile notify.py and install on the i13 yet so tested using a calender event: the popup visible above Communicator 5 but eyegaze actions restricted to communicator itself.  Windows control doesn't apply at this point while c5 is running.  

I think Toast is the the "proper" solution and good for some users, but for accessibility it MUST implement an auto-timeout AND be given alarm priority as the voca must always be set to no notifications, alarms only.  Without implementing both of these, and as Voca software can't handle CLI return info, a small pop-up QT app that just displays the device name only then self-dismisses will be needed.  

>...flag —notify on CLI that if any data returns notify os. If not just return as we currently do.
If you add toast, YES.

willwade, 2022-07-14:
Hmm. Interesting. I assumed it would timeout. Ok. So might need to figure out an alternative. Not sure what that is. I wonder if there is a timeout time setting buried somewhere..

joedevsys, 2022-07-15:
Sorry, I should have been clearer - if notifications have been restricted to Alarms Only (e.g. to avoid being distractions on a VOCA), then the RK notification would need to be an Alarm (scenario attribute) to get through, but it appears Alarms do not timeout and I can't see a way to add one.

So, next thing to try is Focus Assist=Priority Only and configure it to minimise unwanted stuff while permitting RK to send normal notifications. (Your link to the timeout setting noted)

joedevsys, 2022-07-15:
I compiled the notify-py simple example which works ok. Setting  Focus Assist=Priority Only stops it popping up as you'd expect  (the event is shown in the events list ok), however it's not possible to add the test program to the Focus Assist Priority List so the popup's can't be enabled again - the app would need to register an AppUserModelId before it can be added to the priority list.  Not sure if this library supports this. https://github.com/DatGuy1/Windows-Toasts might be better but lacking documentation of this area so going to need some digging.

willwade, 2022-07-17:
Thanks @joedevsys -I'll look into that AppUserModelId - it looks like its more of an installer thing rather than the actual notify thing (https://docs.microsoft.com/en-us/windows/win32/shell/enable-desktop-toast-with-appusermodelid). If thats the case we should be able to register cli-win.exe cli.exe with a AppUserModelId using this https://nsis.sourceforge.io/WinShell_plug-in - but still not 100% convinced. 

By the way - a large amount of commits to the workflow later - you can grab the latest installer on any commits to master on the action tab. e.g. (see at the bottom under artifacts
https://github.com/AceCentre/RelayKeys/actions/runs/2684972767 )

joedevsys, 2022-07-17:
A few layout suggestions:
-Move Change Mode to the device menu instead of a button as it's used infrequently during setup, similar to adding/removing devices.  
-Always restart the timer (not just when not connected) so that if the dongle is disconnected later the change of status is displayed.
-Refresh the current device name on the timer as well so you no longer need a refresh button.

willwade, 2022-07-18:
@joedevsys 

> Thanks @joedevsys -I'll look into that AppUserModelId - it looks like its more of an installer thing rather than the actual notify thing [...]

@joedevsys I've pushed a PR thats WIP to PR #111  - see installer https://github.com/AceCentre/RelayKeys/actions/runs/2692779141 - but I cant see it working. Maybe I'm just looking at focus assist wrong (or more likely - not reading this correctly how this works)

willwade, 2022-07-25:
massive thanks to @f1andrew for his Stellar work on this ! ✊🇺🇦🎉

closed 2022-07-25 by willwade