AceCentre/RelayKeys #112: Make RK/RK win apps available as a priority app in Focus Assist

willwade, 2022-07-19:
**Is your feature request related to a problem? Please describe.**

For users to use AAC - which is typically fullscreen - they will only get notifications if they turn off a bunch of settings. Which is ugly. https://docs.acecentre.org.uk/products/v/relaykeys/guides#a-tip-on-notifications

If it was a "Priority" App it would be a bit nicer.

**Describe the solution you'd like**

Make RK/equivalents a Priority app. We *should* be able to do this using AppSetUserID - I'm trying this with a NSIS extension like seen in this PR https://github.com/AceCentre/RelayKeys/pull/111 - but have no idea if its working. It doesn't look like it. 

as @joedevsys says
>  "I compiled the notify-py simple example which works ok. Setting Focus Assist=Priority Only stops it popping up as you'd expect (the event is shown in the events list ok), however it's not possible to add the test program to the Focus Assist Priority List so the popup's can't be enabled again - the app would need to register an AppUserModelId before it can be added to the priority list. Not sure if this library supports this. https://github.com/DatGuy1/Windows-Toasts might be better but lacking documentation of this area so going to need some digging."

And then me:

> "Thanks @joedevsys -I'll look into that AppUserModelId - it looks like its more of an installer thing rather than the actual notify thing (https://docs.microsoft.com/en-us/windows/win32/shell/enable-desktop-toast-with-appusermodelid). If thats the case we should be able to register cli-win.exe cli.exe with a AppUserModelId using this https://nsis.sourceforge.io/WinShell_plug-in - but still not 100% convinced.

> By the way - a large amount of commits to the workflow later - you can grab the latest installer on any commits to master on the action tab. e.g. (see at the bottom under artifacts https://github.com/AceCentre/RelayKeys/actions/runs/2684972767 )"

See https://github.com/AceCentre/RelayKeys/issues/105#issuecomment-1186463817

**Describe alternatives you've considered**

https://github.com/safing/nsis-shortcut-properties

Not worked either.. Well definitely app is getting registered correctly. But I dont see it in a priority list 

**Additional context**

We could do a popup modal or similar - but it won't work as the AAC app has fullscreen view. It would just get hidden in the background.

joedevsys, 2022-07-23:
The toast notifications are neat apart from the difficulty with Focus Assist.

An alternative for AAC apps could be to have a --copy switch. Put the return message into the clipboard so the app can display in a cell.  e.g. Communicator5 cell script would be something like this to switch put the new device name into the message bar:
    Clear
    Write "Switching device, please wait..."
    Run Program: relaykeys-cli-win.exe cle_cmd:switch
    Delay 8 secs
    Run Program: relaykeys-cli-win.exe cle_cmd:devname --notify
    Paste

It's a bit clunky so getting toast to register properly would be best.

willwade, 2022-07-24:
I think trying it out isn't a bad idea - Funnily enough I thought of another similar idea..
I've added a --copy flag - which means you can do --notify &/or --copy. Would still like to get to the bottom of how you register this properly mind you

joedevsys, 2022-07-25:
Bug in your PR: Line 179 call to copy_return() 2nd parameter command is unknown, should be devcommand.
Suggestion added to the PR.

With this fix --copy flag works really well in Communicator5 using the script in my previous comment. This is an effective alternative, but the user & I both would prefer toast notifications if the Focus Assist can be resolved.

willwade, 2022-07-25:
ok - well im a fan of options.. lets merge it and still keep plugging away at that issue

willwade, 2022-07-26:
FYI http://forums.winamp.com/showthread.php?t=458118

joedevsys, 2022-07-28:
--copy working great after merging modified #121

willwade, 2022-11-09:
This might help - doubt it - but might https://github.com/binwiederhier/ntfy

willwade, 2023-09-14:
Joe - I have a favour to ask. Can you install this app we pulled together. https://github.com/AceCentre/TranslateAndTTS#step-1-download 

Dont worry - you dont really need to set it up. Just add a call in communicator to the exe. 

I want to get it to create an alert. Just getting it offline should do it. Get communicator full screen. 

Does the alert come up with UAC and all that set up? Im kind of incredulous as it is for me.. and I didnt think it would 

If thats the case the way we do notifications (QT with a timeout). We might copy over to relaykeys.. but I don't believe it.