AceCentre/RelayKeys #113: Automate ble_cmd:reconnect when daemon fails to communicate with dongle

joedevsys, 2022-07-19:
ble_cmd:reconnect recovers the comms to the dongle after the dongle is unplugged and the port gets closed.  When the dongle is next plugged in again and this command re-opens the port and resets the dongle.  Comms fail until the user manually initiates sending ble_cmd:reconnect.

It would be useful to automate this when the daemon tries to send a command that fails.

See discussion in #89.

@joeDevSys: 
"I wondered if the reconnect code can be called automatically after a fail and then the quoted command retried but have to avoid endless tries, and I think it would need a considerable delay as it takes the remote some time to accept the reconnection. So maybe best left to the user?? They'll know if the dongles been replugged. 
...
Does ble_cmd:reconnect need adding to the ble loop for wireless mode? Might need slightly different actions?"

willwade, 2022-07-27:
see https://github.com/AceCentre/RelayKeys/pull/123

joedevsys, 2022-07-28:
Auto-Reconnection is working OK now, thanks @f1andrew

closed 2022-07-28 by joedevsys