AceCentre/RelayKeys #115: Improve pop-notification messages

joedevsys, 2022-07-20:
Calling the cli with the --notify switch causes popup (toast) status notifications. The text of these messages is a bit "technical" and not user friendly.

e.g. cli command ble_cmd:devname causes a popup message:
"
Device Command Response
Devname: <MyPhone>
"

A message like this would be better e.g.:
"
Connected to:
<MyPhone>
"

This applies to all end-user messages and notifications generated by RK. 

P.S. Need to be consistent what is meant by "device" in both documentation and connection messages (controller? | dongle? | target?) as it can get a bit confusing.

willwade, 2022-07-21:
Re: Docs. Totally Agree. Its a right old pickle. I've made an initial stab: https://docs.acecentre.org.uk/products/v/relaykeys/#the-relaykeys-ecosystem  

I'm proposing:
- Any reference to RelayKeys-QT => RelayKeys-Desktop (and we will rename this entirely in a PR)
- I regret now we have cli.exe and cli-win.exe. Its a weird hangover from what pyinstaller calls a windowed app or not. I'm tempted to leave as is as it will break to many people's setups now
- I dont like it - but think the device it runs on is the server computer and it connects to "receiving devices". I think we are going to generically call any hardware running the firmware as "RelayKeys Hardware". Doc writing is an art form..

Sure thing on the popups. You are one step ahead of me Joe! I haven't had enough Time to check them all out yet

joedevsys, 2022-07-21:
RelayKeys-Desktop sounds good.
How about calling  receiving devices the "Target Devices"?

willwade, 2022-07-27:
see https://github.com/AceCentre/RelayKeys/pull/123

joedevsys, 2022-07-28:
The pop-up notifications messages look great now, thanks @f1andrew 

This issue can be closed if you're done with docs naming conventions @willwade

closed 2022-07-28 by willwade