AceCentre/RelayKeys #117: Repeat mouse till stop

willwade, 2022-07-21:
Want a way of sending a command mouse movement command (say mousemove:10,10) and repeating that until a user sends another command to stopmove

I wasnt sure best way of doing that. It would all have to be threaded.. 

I think we could hack up a little script to call mousemove on repeat and listen for a keyboard shortcut to stop it..  But this feels.. well a hack..

e.g..

`from pynput import keyboard

def on_activate_stop():
    data = pyperclip.paste()
    converter.say(data)
    converter.runAndWait()
    

def for_canonical(f):
    return lambda k: f(l.canonical(k))

# call mousemove:10,10

hotkey = keyboard.HotKey(
		keyboard.HotKey.parse('<ctrl>+<alt>+s'),
		on_activate_stop)
	with keyboard.Listener(
			on_press=for_canonical(hotkey.press),
			on_release=for_canonical(hotkey.release)) as l:
		l.join()

`

joedevsys, 2022-07-23:
I can see why this is needed. Terminate with something like any other mouse command (e.g. MouseMove 0,0)  rather than a hotkey might feel less hacky?

willwade, 2022-07-24:
yeah. Thinking about it I think this would be better dealt with by the RPC daemon as its constantly running. I just need to remember if its blocking or not.

willwade, 2022-07-25:
So terminating is not that simple - it's easier done by sending a separate app running into the background. I've got a little walkthrough how to do this in the docs. Watch this space.

willwade, 2022-07-25:
See docs now at https://docs.acecentre.org.uk/products/v/relaykeys/developers/relaykeys-cli#command-mousemove-pixelsright-pixelsdown

NB: when we fix #113 and #115 this will be in the latest release that I'll push

closed 2022-07-25 by willwade