AceCentre/RelayKeys #125: Difficulty pairing to i13 for wireless mode

joedevsys, 2022-08-11:
Trying to setup an i13 communication aid for wireless mode but cannot pair it.

Windows 10. 
Itsybitsy nrF52840 Express. 
RelayKeys 3bc6133

Steps taken: 
- Installed RelayKeys and check working in wired mode (ok)
- Launch RelayKeys-QT (Connected to dongle ok by hardware serial.)
- Reset Device List (Device list is cleared)
- Add Device (dongle LED changes from green to  yellow)
- Launch Windows BT settings. Add Bluetooth or other device. Select Bluetooth
- Nothing appears either in RelayKeys-QT or in Windows settings.

Observations:
- I can pair other BT devices to the i13.
- I can setup wireless mode successfully on another Win10 laptop using the same dongle

Further checks:
Performed a BLE Scan from the i13 using **Bluetooth LE Explorer** app from the Windows Store:
Discover and Pair. Start: This lists the RK dongle at it's expected bluetooth address but it has no name (see pic).
[image]
Clicking the Pair button here (with RelayKeys-QT Add Device) does create a pairing and at this point the name RelayKeys appears in both Bluetooth LE Explorer app and in the Windows list of connected devices, and also in RelayKeys-Q. However after creating the pairing in this manner when I try and change mode in RelayKeys-QT, double click the button on the dongle (LED goes blue), and delete relayKeys from the Windows settings it still fails to connect to the dongle. RelayKeys-QT shows Wireless mode, no connection.

I tried deleting registry entry for the dongle's BT address from HKEY_LOCAL_MACHINE > SYSTEM > CurrentControlSet > Services > BTHPORT > Parameters > Devices and tried again but no change.

Using HCITool on a Linux machine to do an LE scan:
$sudo hcitool -i hci0 lescan
{dongleBTaddress} (unknown)
{dongleBTaddress} RelayKeys
{dongleBTaddress} (unknown)
{dongleBTaddress} RelayKeys
{dongleBTaddress} (unknown)
{dongleBTaddress} RelayKeys
...

This seems to show the dongle returning two scan responses, one of them has no name. Is this correct? I don't see it with other devices.

As I can pair on another machine I assume this is a rogue machine, but the "unknown" response from the dongle might be contributing?

willwade, 2022-08-11:
Hmmm. This is reminiscent of a problem we see sometimes on iOS. Like it sees devices with two different names. Needs a bit of digging around. Thanks Joe for the detailed notes! 

@f1andrew

f1andrew, 2022-08-12:
I have same thing that when scanning with hcitool I can get two type of entries from dongle, with name and without name.
I tried Adafruit's minimal ble periheral example(ble_uart), and I get same result when scanning. So it can depend on how Adaftuit core setting up bluetooth and etc.

I don't know why communicator can't see dongle in bluetooth setting while other devices can.
But you were able to connect to it with external Bluetooth app and added communicator to device list.

There might be problem with bleak library on communicator, but it needs to be confirmed.
Here is isolated example for testing how bleak works on communicator.
Python script tries to find relaykeys dongle and after that connects to it and prints its services.
Here is compiled version with pyinstaller. 
Compiled version should create console window and put all prints here.

Let me know does it work on communicator.
[relaykeys_connect.zip](https://github.com/AceCentre/RelayKeys/files/9316503/relaykeys_connect.zip)

joedevsys, 2022-08-31:
On a different laptop Relaykeys_Connect.exe starts scanning and then (after QT Add Device) finds the dongle, reports "RelayKeys discovered <address>" then lists services ok.  Windows BT control panel offers RelayKeys for pairing.

On the i13 computer it reports "scanning... scanning...  scanning..."  but never shows the dongle. Windows BT control panel pairing never  lists RelayKeys. 

@willwade are you able to try this on another i13?

willwade, 2022-09-01:
Yeah I will for sure Joe. I’ve been off for a bit but should have some time Monday to dig into this. Thanks

willwade, 2022-09-07:
Ok - My first attempt in this is that something up with my i13+ - or something in general up. I cant see relay keys even in wired mode in the bluetooth scanning screen on Windows. It does state the i13 is using a BLE bluetooth module - https://www.tobiidynavox.com/pages/product-support-i-series?tab=3 - so its not its a classic problem. If I plug in a 3rd party BLE dongle in - it does appear. Somethings v odd..

joedevsys, 2022-09-07:
Thanks for confirming that. So now we know multiple i13's exhibit the same behaviour, I suspect it's specific to the i13 as I can get an HP laptop and various android devices to pair.

willwade, 2022-09-08:
Ok - @f1andrew  - there's something about the way relaykeys is advertising itself. I cant figure it out. Basically our MorAce (which we can run on the same board) - is being seen by this device. But RelayKeys isn't. I can't figure out how we fix this. My only mad idea is to convert our relaykeys code to circuitpy! but that cant be right.. ! :)

joedevsys, 2022-09-08:
The ble_keyboard example gets listed for pairing. Restoring the latest RelayKeys code it does NOT get listed. (I restarted Windows first to clear any recently seen MAC addresses and names).

I clear the pairings each time so I am always trying to make the first pairing and no HID connection in progress.

willwade, 2022-09-12:
OK - so turns out if we comment out this line - it works. 

https://github.com/AceCentre/RelayKeys/blob/eb7a11f6a929114b613832dd95ab8cc7398a2d3b/arduino/arduino_nRF52840/arduino_nRF52840.ino#L278

@joedevsys - if you get a mo - mind double checking?

joedevsys, 2022-09-13:
Commenting that line allows the i13 to list "ItsyBitsy nrf52840 Express" and pair to it. The i13 is now listed in the dongle device list.

After that I still cannot connect in wireless mode (QT switch mode to BLE serial, double press dongle button for blue LED, remove ItsyBitsy from device list--> no connection). I presume the daemon is still searching for the name "relaykeys" which is no longer advertised.

willwade, 2022-10-11:
Im not liking closing this because I dont feel we really resolved "WHY" the i13 couldn't see that name.. But heck - thats life - we have a fix it seems

closed 2022-10-11 by willwade