AceCentre/RelayKeys #127: Add command to switch Arduino modes

joedevsys, 2022-09-24:
At the moment you can switch the daemon mode with the "relaykeys-cli.exe daemon:switch_mode" command, but the only way to change the dongle mode is by physically pressing the user switch. This makes it awkward to encapsulate the dongle. 

How about adding a "AT+changemode" command to the Arduino code and a matching relaykeys-cli.exe daemon:switch_dongle_mode" command to the CLI?

willwade, 2022-09-24:
I guess that would make sense. Would we need a second command to read what mode it’s in?

joedevsys, 2022-09-24:
Probably not, maybe the LED colour is sufficient.

willwade, 2022-10-03:
@joedevsys have a look at that PR. we made a bit of a design decision on this one. If you send the command in wired mode. It should work fine. But then it will disappear if it’s not been setup correctly. We could have rewritten everything to continue to monitor the usb serial bus bit to be honest it felt like a lot of work. Let us know if this does the trick for you.

closed 2022-10-11 by willwade