AceCentre/RelayKeys #131: Keys that are not valid

willwade, 2022-10-25:
At https://docs.acecentre.org.uk/products/v/relaykeys/using-relaykeys/relaykeys-cli - under the Keys dropdown we have keys such as: LOCKING_CAPSLOCK
LOCKING_NUMLOCK
LOCKING_SCROLLOCK

Do these work? Are any other keys spurious (Of course we had vol keys which now do work..)

f1andrew, 2022-10-26:
LOCKING_CAPSLOCK
LOCKING_NUMLOCK
LOCKING_SCROLLOCK
Are defined in hid keyboard codes(https://learn.adafruit.com/introducing-adafruit-ble-bluetooth-low-energy-friend/ble-services#hid-keyboard-codes-2435765), but seems they don't do anything.

Same situation with:
    ("EXECUTE", 0x74),  # Keyboard Execute
    ("AGAIN", 0x79),  # Keyboard Again
    ("ALTERASE", 0x99),  # Keyboard Alternate Erase
    ("ATTENTION", 0x9A),  # Keyboard SysReq/Attention
    ("CANCEL", 0x9B),  # Keyboard Cancel
    ("CLEAR", 0x9C),  # Keyboard Clear
    ("PRIOR", 0x9D),  # Keyboard Prior
    ("RETURN", 0x9E),  # Keyboard Return
    ("SEPARATOR", 0x9F),  # Keyboard Separator
    ("OUT", 0xA0),  # Keyboard Out
    ("OPER", 0xA1),  # Keyboard Oper
They defined in USB standard, but probably not supported by most operating systems. I haven't seen keyboards with such keys

f1andrew, 2022-10-26:
Keys:
("STOP", 0x78),  # Keyboard Stop
("POWER", 0x66),  # Keyboard Power

Are working for android but not for iOS. Probably, because iOS can receive them only as media keys.

willwade, 2022-10-26:
Ok. Thanks for that. I'll just update the docs