AceCentre/RelayKeys #142: Support multilingual docs

quaxalber, 2023-08-04:
**Is your feature request related to a problem? Please describe.**
The docs are currently English-only. 

**Describe the solution you'd like**
According to GitBook multi-languages can be implemented fairly straightforward by adding a sub-directory for each language and a LANGS.md file.

willwade, 2023-08-04:
For sure. What languages? And before we do this - have you got any recommendations to the docs? I’m more than aware they aren’t great 

if you like I can add you to our gitbook project and you can make revisions - or do it via a PR

Many thanks

quaxalber, 2023-08-04:
Great! As a native speaker, I'd like to translate the docs to German. Happy to accept your invite. I do have a few suggestions in mind wrt the English version.

willwade, 2023-08-04:
Great! Ping me your email. Find my email here https://github.com/willwade

quaxalber, 2023-08-06:
Perhaps the way to go is using collections. Check this: https://docs.gitbook.com/publishing/share/collection-publishing

gavinhenderson, 2023-08-07:
Hey 👋  @quaxalber multi-lingual docs would be really helpful.

Im a colleague of Will, who is off for a few days, so just picking up where he left of.

Collections does seem to be the most sensible way to pull off multilingual docs. The only slight issue is that we already use collections. You will notice that when you have the relay keys docs it allows you to jump to other Ace Centre products documentation.

What I will do is create a new 'space' in the current collection, then I will rename the current docs to RelayKeys (english) and the new one to RelayKeys (german).

If we end up with lots of translations I will move it out of the current collection and do something longer term but for now this will work.

This is the new space I made, as you can see I have set it up as empty: https://docs.acecentre.org.uk/products/v/relaykeys-german/

@quaxalber I have now added you as an editor to the new space so feel free to edit away and submit a pull request on GitBook. Thanks again for your contribution!

willwade, 2023-08-10:
@quaxalber  I've reformatted our English docs - let me know your thoughts. Ive just moved some pages around to make it easier to follow and understand. There are still some missing parts like firmware uploading for the receiving dongle - but I feel its in a better shape

quaxalber, 2023-08-10:
@gavinhenderson Thx for the follow-up and for setting this up! 

@willwade Impressive rework! I really like it. Much clearer structure. Once you did the initial translation, I will tweak as necessary

quaxalber, 2023-08-14:
Great stuff. A good place to start.

willwade, 2023-08-16:
I've managed to auto translate the rest. NB: Using DeepL 
I'm going to close this for now - but feel free to open any issues you have :)

closed 2023-08-16 by willwade