AceCentre/RelayKeys #143: Revisit GridSet

quaxalber, 2023-08-04:
**Is your feature request related to a problem? Please describe.**
The current [GridSet](resources/aac-software/RelayKeys.gridset) has some room for improvement: 

**Describe the solution you'd like**

- [ ] _Fix key commands_
Some keys are mapped to a wrong command, likely due to copy paste, e.g. the `+`/`-` keys are mapped to `keypress:2` and `F6`-`F8` are mapped to `keypress:F5` in `RelayKeys - Keyboard`. 

- [ ] _Split up `RelayKeys - Keyboard`_
While it's a nice idea to have everything in one grid, this results in very small buttons and frequent misclicks, especially using Eyegaze. I suggest to move some of the functionality into separate grids such that the buttons may be enlarged. 

- [ ] Use `type` instead of `keypress` 
I think on the frontend one should not be concerned with HID mappings. That is to be done by the respective keymap file. Instead, the command ought to be just `type:&` for instance. As a side effect much of the grid may be reused when creating new keyboard layouts for different keymaps.

willwade, 2023-08-10:
I hear you! Its on the to-do list

quaxalber, 2023-08-10:
Cool. I am currently tinkering with a layout myself. Will share my draft once finished. Is there a way to hold down buttons such as LSHIFT or LCTRL?

quaxalber, 2023-08-14:
@willwade Curious for your feedback. Would adapt the GridSet for US and UK accordingly

commit 789eafda (2023-08-15): Revisited GridSet for DE (closes #143)
closed 2023-08-15 by willwade