AceCentre/RelayKeys #144: Last used modifier remains active on recipient

quaxalber, 2023-08-08:
**Describe the bug**
When sending key strokes through RelayKeys, those modifiers of the last character remain active on the recipient until new key strokes are sent via RelayKeys. 

**To Reproduce**
Steps to reproduce the behavior:
1. Send key strokes with modifiers such as capital letters, e.g. using the `type` command 
2. Type further characters on the recipient without RelayKeys, e.g. using a hardware keyboard 
3. Observe that all characters retain the last modifier sent through RelayKeys 

**Expected behavior**
Modifiers should deactivate unsless explicitly toggled on

**Screenshots**
Video Demo: 
https://github.com/AceCentre/RelayKeys/assets/64684396/f4d6f309-dab7-4ade-b418-9df0087e3fe6

**Desktop (please complete the following information):**
 - OS: Windows 11 Version 22H2 (Build 22621.2070)

**Additional context**
- RelayKeys v1.99.7

willwade, 2023-08-09:
Thanks for your great video - hmmm. I'll look into this

willwade, 2023-08-14:
@quaxalber - I've pushed a change made by @f1andrew  that attempts to fix this. It simply releases the keys after a paste or a type command via the CLI.  as @f1andrew states "this issue may manifest if user didn't deactivate modifier explicitly after using "keyevent" and "keypress",
because modifiers are latched(https://github.com/AceCentre/RelayKeys/issues/87)"

I'll do release if that looks ok @quaxalber

quaxalber, 2023-08-14:
This looks good. Infact, I had pretty much the same idea in mind when looking through the code. However, when working on #143, I was using `keyevent` to latch modifiers such as LSHIFT and I noticed the same behavior. It wasn't sufficient to simply send `keyevent:LSHIFT,,0`. I had to send another `keyevent` in order for the unlatch to register. In this case I am sending an "empty" `keyevent` afterwards which does the trick (`keyevent:,,0`).

My educated guess is that the issue lies within [blehid_send_keyboardcode](https://github.com/AceCentre/RelayKeys/blob/master/blehid.py#L233). I just haven't familiarized myself enough with code to pinpoint it.

f1andrew, 2023-08-15:
For now, use "keyevent:LSHIFT,,0" to remove remaining modifiers. A bit later there will be fix for this.

quaxalber, 2023-08-16:
@f1andrew Works like a charm! And you even read my mind and implemented a [release_keyboard](https://github.com/AceCentre/RelayKeys/pull/146/files#diff-5897c8b554795b3c2e4f6bf7ffd42226dd1b2796920f172a8899fc66b18c2035R594) command. Great stuff

willwade, 2023-08-16:
> @f1andrew Works like a charm! And you even read my mind and implemented a [keyboard_release](https://github.com/AceCentre/RelayKeys/pull/146/files#diff-5897c8b554795b3c2e4f6bf7ffd42226dd1b2796920f172a8899fc66b18c2035R594) command. Great stuff

Awesome. Thanks both! 
merged. I’ll update the docs with this then do a release

closed 2023-08-16 by willwade