AceCentre/RelayKeys #148: Pairing issue on dual boot after switching OS

quaxalber, 2023-08-19:
**Describe the bug**
After switching to the other OS in my dual boot setup (Win 11 & Manjaro Linux) the Bluetooth Pairing to AceRK breaks. After removal and re-pairing everything is back to normal. This doesn't happen after normal reboots without switching OS. I believe this is an issue of RK as it happens on either OS. 

**Screenshots**

https://github.com/AceCentre/RelayKeys/assets/64684396/49120337-0600-49f0-93d2-b0c48d762f2a

[image]

**Desktop:**

- OS 1: Windows 11 Version 22H2 (Build 22621.2070)
- OS 2: Manjaro Linux (Kernel: 6.4.9-1-MANJARO)

Additional context

- RelayKeys v1.99.7
- [nRF52840 USB Key with TinyUF2 Bootloader](https://www.adafruit.com/product/5199)

willwade, 2023-08-20:
@f1andrew this video looks similar to that bug we had which you fixed - last time it was on windows. Is it the same thing again?

quaxalber, 2023-08-20:
I'm happy to provide further information as required

f1andrew, 2023-08-20:
This is common problem related to dual boot. Here you can read more about it: 
https://unix.stackexchange.com/questions/255509/bluetooth-pairing-on-dual-boot-of-windows-linux-mint-ubuntu-stop-having-to-p

In general you need to synchronize pairing keys on both OS.
There are many posts and their solutions differ, so I can't pick one that will work for sure.
You need to find instructions that are close to your environment(Windows 11 + Manjaro) and try them.

quaxalber, 2023-08-20:
Gotcha! Fixed it with [this](https://github.com/gawells/dual-boot-bluetooth-pair). Thx for pointing me in the right direction @f1andrew

closed 2023-08-20 by quaxalber