AceCentre/RelayKeys #149: Expand language support

quaxalber, 2023-08-20:
**Is your feature request related to a problem? Please describe.**
Currently English (UK and US QWERTY) and German (QWERTZ) are supported for Grid 3. 

**Describe the solution you'd like**
I suggest to revisit the existing layouts for other AAC platforms. Also I think adding further layouts would be beneficial to the community and the adoption of RelayKeys. As a first step, adding the layouts only should be sufficient since probably most people interested in using RelayKeys will manage with the English docs. Thus localized docs may be added at a later point in time. 

**Describe alternatives you've considered**
According to ChatGPT these are the most commonly used keyboard layouts in Europe and North America:
https://chat.openai.com/share/9c95569b-6d94-423a-ac9d-973bd1211f70

Do you have any specific layouts in mind that should be implemented first? I'd take care of the Grid side of things. 
Do you have users outside of Europe and North America you are aware of?

willwade, 2023-10-05:
I've been wondering about this. I did some hacking with the standard layout system of linux to bulk convert them to our JSON format. Its far from done - and I may never finish this - but it might be useful for new layouts we want to use. Take a look at https://acecentreuk.sharepoint.com/:f:/s/AnonymousShares/EmtqWZU9y7xNsEppKb0NnpoBqaadNRziB67kh446X9ybUA?e=N7dWG0

Note the python script in there for doing the conversion. output_json holds the json output

quaxalber, 2023-11-04:
Hey @willwade, my apologies I wasn't able to reply earlier. The SharePoint links appears to have expired. Did you follow this approach further? 

However, I think I may have found a different path to include many more layouts with reasonable effort: 
Check out https://github.com/Neradoc/Circuitpython_Keyboard_Layouts

The repo contains quite a few layouts, ready to be used with [adafruit_hid](https://github.com/adafruit/Adafruit_CircuitPython_HID). These were extracted from the windows drivers and should therefore be complete. The best part is, they provide a [web interface](https://github.com/adafruit/Adafruit_CircuitPython_HID) to generate any of the [210 layouts](https://kbdlayout.info/) shipped with windows at the push of a button. 

While this would necessitate some refactoring to migrate from the json keymaps to adafruit_hid, it strikes me as the best approach moving forward. What are your thoughts?

willwade, 2023-11-04:
Hmmm. That is neat. I can see a conversion tool to output from that to our layout. 

Nb. Originally we wanted to use circuitpy for relaykeys. The reason we didn't is you couldn't change MAC address. I think we need to investigate this. I also think we might be able to improve the daemon/serial idea if we stepped back and redesigned things . I think that's a source of some weirdness people find. Well that and no feedback loop which is tough to solve.

willwade, 2025-06-09:
https://github.com/AceCentre/RelayKeys/blob/master/KEYBOARD_LAYOUTS_IMPLEMENTATION.md

closed 2025-06-23 by willwade