AceCentre/RelayKeys #153: Automatically send notifications when new target sucessfully connected

joedevsys, 2023-09-14:
When changing target device using  ble_cmd:switch or  ble_cmd:switch="Devname" (etc) there is no notification when the target finally makes a successful connection. It is necessary for the user or his user-level application to poll at some later time to display the new status. Typically this might need to be 8-10 seconds later.

Unfortunately scripting in AAC software is limited and this is tricky. In Communicator 5, for example, it's possible to add delay commands and then call relaykeys-cli.exe ble_cmd:devname, however the delay blocks the current button and seems to cause UI issues (possibly threading issues?). At the moment the most reliable method is the user to wait 10 seconds and *manually* use a second button to poll the status. This works but isn't great. 

Is it possible to trigger a thread in the cli that waits 8-10 seconds and then auto-runs ble_cmd:devname --notify? 
Alternatively a permanent thread could auto-run ble_cmd:devname --notify every time a successful connection is made?

f1andrew, 2023-09-16:
@joedevsys
Sorry for delay.

Yes, we can add automatic polling right into cli "switch" command.
You said that there are problems with delays in AAC software.
Does while loop with some delay in python code will work this case or it's needed to start thread that will be running in background and not block shell?

joedevsys, 2023-09-16:
Don't block the initial call to cli switch. Start a background thread and return.

f1andrew, 2023-09-18:
I've added automatic polling of device name in same pull request : https://github.com/AceCentre/RelayKeys/pull/152
If switch command was run with --notify flag then poll_devname process will start on background.
It will be running until device name changes from "NONE" and send notification with new devname or it will just stop running after timeout(40 seconds now).

Let me know how it works.

willwade, 2023-10-05:
I think the bugs are largely gone in this build. I'm leaving it as pre-release for a week or so. If anyone gets a chance to check it out - https://github.com/AceCentre/RelayKeys/releases/tag/2.01 Updated the docs. Going to close this - but someone feel free to reopen if you spot anything. Thanks again everyone :)

closed 2023-10-05 by willwade