AceCentre/RelayKeys #155: [Request] New RK Release

quaxalber, 2025-01-22:
Hey Will and team, 

I hope you are doing fine. 

I noticed there are quite a few [commits since the latest release](https://github.com/AceCentre/RelayKeys/compare/2.01...master). When do you plan for the next release? 

Best
quaxalber

quaxalber, 2025-05-11:
Hey @willwade 

Any chance you could squeeze in a release? :)

willwade, 2025-05-12:
Oh go on then. Any additional things you need ?

willwade, 2025-06-08:
Hi @quaxalber! 👋

Thanks for your patience and for requesting the new release. I've just created **Release 2.02** which includes all the changes that have been made since 2.01.

🎉 **[RelayKeys 2.02 is now available!](https://github.com/AceCentre/RelayKeys/releases/tag/2.02)**

## What's included in 2.02:
- Device polling improvements (`poll_devname.py`)
- Better notification system for device switching
- Documentation updates and new guides
- Various stability improvements

The release includes all commits since September 2023, so you should have all the latest improvements and fixes.

Thanks for keeping us on track with releases! 🚀

closed 2025-06-08 by willwade

quaxalber, 2025-06-15:
Thanks so much for this! :)

willwade, 2025-06-15:
No problem. I'm away currently. Back in a week or so. I'll do another release signed then and with the other features I've been working on

quaxalber, 2025-06-21:
Hi @willwade 

I just noticed that 2.02 doesn't contain the installer nor the UF2 files. Would you mind adding those? Thanks!

willwade, 2025-06-21:
Yeah. No I realised that. I closed it not releasing it. I intend to get that done tomorrow! And get more release out too

willwade, 2025-06-23:
So check out https://github.com/AceCentre/RelayKeys/releases/tag/2.03 - Im going to find a esp32 device for the esp32 firmware.. apologies for the delay!

quaxalber, 2025-06-29:
Thanks so much for this @willwade 
Unfortunately the daemon won't start with the latest release 

[image]

quaxalber, 2025-06-29:
No worries at all. I appreciate your quick response!

quaxalber, 2025-08-01:
Hey Will, any chance you could squeeze this in? Thanks!

quaxalber, 2025-08-27:
*ping* :)

willwade, 2025-09-18:
No nudge away! Each time I sit down to do this I start but I'm finding some horrible bugs in things. I might roll back. Release. Then work on fixes. Maybe Friday I'll do this!